STMicroelectronics/STM32CubeL5 #1: In project tree, in folder ..\CMSIS_RTOS_V2\ cmsis_os.c file is missing

wojciech934, 2020-02-25:
Hi
I use free rtos via STM32CUBE towards H7 target w/cmsis v2 option. When I set up free rtos and just create some tasks, then project is compilable, I see tasks are running. When I want to additionally use signals, then after configuration and placing calls osSignalWait(), osSignalSet(),  the project is not linkable anymore. These definitions are missing. In cmsis_os2.c they are not defined so I suppose that cmsis_os.c should be available in the project tree for CMSIS V2. Alternatively cmsis_os2.c should be updated by these definitions.

ALABSTM, 2020-04-03:
Hi @wojciech934,

Thank you for this report. May I ask you for some question however to better understand the issue?
* According to what you wrote, you use a H7 target. However, you created this issue in the frame of the L5 firmware repository. Would you mind precising the reference of the microcontroller and the board you are using?
* Would you also mind precising the version of the firmware you are using? This is in order to ensure you are using the latest one published.
* The "CMSIS V2" you are referring to is the [CMSIS_RTOS_V2](https://github.com/STMicroelectronics/STM32CubeL5/tree/master/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2) folder, is it not?

Thank you in advance for your answers.

With regards,

ALABSTM, 2020-04-14:
Hi @wojciech934,

Any feedback on the questions above?

Thank you,

ALABSTM, 2020-04-15:
Hi @wojciech934,

I submitted your question to our development teams. They said `osSignalWait()` and `osSignalSet()` are **deprecated** functions ([link](http://www.keil.com/pack/doc/CMSIS_Dev/RTOS2/html/os2MigrationFunctions.html#mig_signalEvents)). New functions shall be used instead ([link](http://www.keil.com/pack/doc/CMSIS_Dev/RTOS2/html/os2MigrationFunctions.html#mig_threadFlags)).

I hope this answers your question. There are no missing files in conclusion, simply deprecated functions that shall not be used any more.

With regards,

ALABSTM, 2020-06-05:
Hi @wojciech934,

I hope you are doing well. Any feedback after my last comment?

With regards,

ALABSTM, 2020-07-22:
Hi @wojciech934,

Please allow me to close this issue as no feedback has been received. Thank you for your comprehension.

With regards,

closed 2020-07-22 by ALABSTM