STMicroelectronics/STM32CubeL5 #2: Trusted Firmware-M for Nucleo L552ZE-Q

berthob98, 2020-06-09:
Hello @ALABSTM and development team,

we are currently trying to port trusted firmware to the Nucleo L552ZE-Q board. Our starting point is the TF-M project working on the L562E-DK board. Do you have information if this works and which changes are needed on the L562E-DK project to make it work on the Nucleo L552ZE-Q board?

Thank you

ALABSTM, 2020-06-23:
Hi @berthob98,

Please excuse this somewhat late reply. Your question will be transmitted to our development teams. I will be back to you as soon as they provide me with the answer.

In the meanwhile, may I ask which IDE are you using (in case some aspects are IDE-dependent)?

With regards,

berthob98, 2020-06-23:
We are using STM32CubeIDE

ALABSTM, 2021-06-17:
Hi @berthob98,

I apologize again for this late reply. This support channel being dedicated to report issues or propose enhancements and fixes related to the software provided within this repository.

For any other type of support like porting a project from one board to another, please submit a post to the **ST Community** on the [STM32 MCU](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) dedicated page.

Sorry again for the late reply. Please allow me to close this issue now.

With regards,

closed 2021-06-17 by ALABSTM