STMicroelectronics/STM32CubeL5 #3: Typo in Projects/NUCLEO-L552ZE-Q/Applications/SBSFU/SBSFU_Loader/STM32CubeIDE/postbuild.sh

Poohl, 2021-02-22:
# Setup
* STM Cube IDE v 1.5.0 on PopOS 20.04 LTS (based on Ubuntu)
 * NUCLEO-L552ZE-Q board (not really relevant)

# Description
when building the Secure Loader Project, the postbuild python script fails with a "file not found" error when opening `secure/Release/SBSFU_Loader_Secure.bin`

```
Traceback (most recent call last):
  File "<truncated>/STM32CubeL5/Middlewares/Third_Party/mcuboot/scripts/imgtool.py", line 417, in <module>
    imgtool()
  File "/usr/lib/python3/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "<truncated>//STM32CubeL5/Middlewares/Third_Party/mcuboot/scripts/imgtool.py", line 365, in ass
    with open(file, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: './../../secure/Release/SBSFU_Loader_Secure.bin'
```
# Reproduce
1. clone this repo
2. make all .sh scripts executable
2. open the STM32CubeIDE
3. open projects from filesystem, choose `STM32CubeL5/Projects/NUCLEO-L552ZE-Q/Applications/SBSFU`
4. compile all projects as described in `STM32CubeL5/Projects/NUCLEO-L552ZE-Q/Applications/SBSFU/readme.txt`
5. the last step, the Loader NonSecure project will fail.

# FIX
in `Projects/NUCLEO-L552ZE-Q/Applications/SBSFU/SBSFU_Loader/STM32CubeIDE/postbuild.sh` capitalize the s in secure in line 8:
```
- loader_s=$projectdir"/../../secure/Release/SBSFU_Loader_Secure.bin"
+ loader_s=$projectdir"/../../Secure/Release/SBSFU_Loader_Secure.bin"
```

RKOUSTM, 2021-03-12:
Hi @Poohl,

Thank you for your contribution. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-03-12:
ST Internal Reference: 103035