STMicroelectronics/STM32CubeL5 #6: HAL_Delay() does not count timer ticks

Malek44, 2021-05-31:
**Set-up**
 * Board: STM32L562E-DK
 * IDE: All.

**Description**
The implemented HAL_Delay(), [here](https://github.com/STMicroelectronics/STM32CubeL5/blob/master/Projects/STM32L562E-DK/Applications/FatFs/FatFs_uSD_TrustZone/Secure/Src/main.c#L346) and [here](https://github.com/STMicroelectronics/STM32CubeL5/blob/master/Projects/STM32L562E-DK/Applications/FatFs/FatFs_uSD_TrustZone/NonSecure/Core/Src/main.c#L267), does not appear to use the timer tick to count for the delay supposedly in milliseconds.

ASELSTM, 2021-06-08:
Hi @Malek44,

The issue you pointed out is confirmed, an internal bug tracker has been logged and a fix will be implemented and made available in the frame of a future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-06-08:
ST Internal Reference: 108588

ASELSTM, 2023-02-17:
Hi @Malek44,

The issue you reported has been fixed in the frame of version v1.5.0 of the STM32CubeL5 published on GitHub.

Thank you again for having reported.

With regards,

closed 2023-02-17 by ASELSTM