STMicroelectronics/STM32CubeL5 #7: PX4 Security SIG Project

korkeep, 2021-06-14:
Hi 😀

I am working on my project of ensuring the hardware security of the Drone's Flight Controller.
I am planning to use **ARM PSA** to ensure security of Drone System at hardware level.

I would like to contribute to **[STM32CubeL5](https://github.com/STMicroelectronics/STM32CubeL5)** by using **NUCLEO-L552ZE-Q**.
**ARM PSA** is hardware security architecture creating two environments that can run simultaneously on a single core:
**Secure State** and **Non-Secure State**.
My project Link: https://github.com/korkeep/PX4-Trust

How can I contribute to the project, or are there any restrictions? 🤔
If you have any related reference, please send it to korkeep@naver.com.

Thank you sincerely 🙏

RKOUSTM, 2021-06-24:
Hi @korkeep,

Thank you for your suggestion. Your request has been forwarded to our development teams. We will be back to you as soon as we get an answer. 

Thank you again for you suggestion.

With regards,

korkeep, 2021-06-28:
I am referring to the SBSFU of STM32CubeL5.
I am developing drone security system by using **NUCLEO-L552ZE-Q**, and there are two issues currently.

1. If I want to **apply Secure Boot for a new binary file** (Ex: px4.bin), which part of the source code should be changed?
2. How to **turn off Secure Boot and return to the original state**? Using regression.sh doesn't bring it back. 😢

Thank you sincerely 🙏

RKOUSTM, 2021-06-30:
Hi @korkeep, 

Thank you for your contribution.

Unfortunately, our development teams have not shared a final decision about your request. It may take sometimes to answer. I will be back to you as soon as they provide me with their answer. We count on your patience and comprehension.

Concerning the Secure Boot point, you can referred to the application note [AN5447](https://www.st.com/resource/en/application_note/dm00684243-overview-of-secure-boot-and-secure-firmware-update-solution-on-arm-trustzone-stm32l5-series-microcontrollers-stmicroelectronics.pdf) "Overview of Secure Boot and Secure Firmware Update solution on Arm® TrustZone® STM32L5 Series microcontrollers" and user manual [UM2671](https://www.st.com/resource/en/user_manual/dm00678763-getting-started-with-stm32cubel5-tfm-application-stmicroelectronics.pdf) "Getting Started with the TF-M applications" available from the STMicroelectronics microcontroller website www.st.com. Also, you can use the application [TFM](https://github.com/STMicroelectronics/STM32CubeL5/tree/master/Projects/STM32L562E-DK/Applications/TFM) and [SBSFU](https://github.com/STMicroelectronics/STM32CubeL5/tree/master/Projects/NUCLEO-L552ZE-Q/Applications/SBSFU) proposed by ST all of them are working well.

The duplicated issue #8 will remain closed. This request will be kept open to track the reported point.

Thank you again for your contribution.

With regards,

RKOUSTM, 2021-08-31:
Hi @korkeep,

First, allow me to thank you for your contribution. Our development teams said your proposal could not be taken into consideration as we don't accept aspect related to Customer project contributions in our STM32 GitHub repositories, only requests for information and suggestions to fix a problem.

Please, allow me to close this issue. Thank you once again for your contribution. We are looking forward to reading from you again.

With regards,

closed 2021-08-31 by RKOUSTM