STMicroelectronics/STM32CubeL5 #8: Drone Security Project: SBSFU Implementation

korkeep, 2021-06-28:
Hi 😀
I mentioned earlier that I am working on drone security project.

For this, I am referring to the **SBSFU** of STM32CubeL5.
I am developing drone security system by using **NUCLEO-L552ZE-Q**, and there are two issues currently.

1. If I want to **apply Secure Boot for a new binary file** (Ex: px4.bin), which part of the source code should be changed?
2. How to **turn off Secure Boot and return to the original state**? Using regression.sh doesn't bring it back. 😢

If you have any related reference, please send it to korkeep@naver.com.

Thank you sincerely 🙏

RKOUSTM, 2021-06-30:
Hi @korkeep,

Thank you for your contribution.

This request will be forwarded to our development teams. I will be back to you as soon as they provide me with their answer.

This duplicated issue will remain closed. Your reported issue [#7](https://github.com/STMicroelectronics/STM32CubeL5/issues/7) will be kept open to track the point.

Thank you again for your contribution.

With regards,

RKOUSTM, 2021-06-30:
Duplicate of #7.

closed 2021-06-30 by RKOUSTM