STMicroelectronics/STM32CubeL5 #9: USB Middleware is not working for L562

tickelton, 2021-07-30:
**Describe the set-up**
 * Custom STM32L562CET board
 * Same behavior observed on STM32L562E-DK discovery board, but I no longer have access to that board
 * STM32CubeIDE 1.6.0 and 1.7.0, GNU Tools for STM32 (9-2020-q2-update)

**Describe the bug**
The USB Middleware is not working correctly for L562.
The behavior is identical for CDC, HID, and mass storage classes (others were not tested).
The USB device is not correctly recognized by either Windows or Linux.
Here's the syslog output on Linux:

kernel: usb 3-1: new full-speed USB device number 38 using xhci_hcd
kernel: usb 3-1: device descriptor read/64, error -71
kernel: usb 3-1: device descriptor read/64, error -71
kernel: usb 3-1: new full-speed USB device number 39 using xhci_hcd
kernel: usb 3-1: device descriptor read/64, error -71
kernel: usb 3-1: device descriptor read/64, error -71
kernel: usb usb3-port1: attempt power cycle
kernel: usb 3-1: new full-speed USB device number 40 using xhci_hcd
kernel: usb 3-1: Device not responding to setup address.
kernel: usb 3-1: Device not responding to setup address.
kernel: usb 3-1: device not accepting address 40, error -71
kernel: usb 3-1: new full-speed USB device number 41 using xhci_hcd
kernel: usb 3-1: Device not responding to setup address.
kernel: usb 3-1: Device not responding to setup address.
kernel: usb 3-1: device not accepting address 41, error -71
kernel: usb usb3-port1: unable to enumerate USB device

Since the error messages start as soon as MX_USB_Device_Init(); is called, it looks like the USB controller is initialized and powered correctly, but interrupt handlers are not installed properly or do not get called at all so that the device does not respond to requests by the host controller.

**How To Reproduce**
Create a basic project on L562 with USB Middleware for a CDC or HID device.
This should at least result in the device being enumerated correctly on the host, but instead only the above errors occur.

**Additional context**
This seems to be the same problem as reported here:
https://community.st.com/s/question/0D73W000000I56ZSAS/detail

An identical setup for F302 using STM32CubeF3 works flawlessly.

On the L562 board, when forcing the controller into DFU boot mode using either the BOOT0 pin or by jumping to system memory from the application software, USB works correctly. The device is enumerated by the host controller, and a firmware update via DFU works perfectly.

tickelton, 2021-08-02:
After some more debugging I was able to solve this issue, at least for my use case.
I believe this is not an STM32CubeL5 issue after all.

The problem seems to be the Clock Issue Resolver of CubeMX, which generates an invalid configuration for L562 (probably L5xx) boards without HSE clock.

For STM32L562E-DK and other L5 boards without external clock source, the resolver will generate a configuration where both the system clock, as well as the usb clock (indirectly via PLLSAI1Q) are driven by MSI.

Apparently (and as suggested by the datasheets) MSI doesn't generate a sufficiently precise clock signal to drive the USB controller.

Setting the USB clock source to HSI48 and the system clock source to HSI solves the problem and results in a working configuration where all USB examples run without problems out of the box.

On another custom board with external 16MHz oscillator that drives both the system and usb clocks the issue didn't occur at all because the clock configuration was valid from the start.

I will leave this issue open in case I'm mistaken and this requires a fix in STM32CubeL5.
I this isn't the case and this is purely concerns CubeMX, feel free to just close this issue.

RKOUSTM, 2021-08-06:
Hi @tickelton,

Thank you for this report. The point you reported is related to the CubeMX generation problem and not directly related to the firmware published in this repository. Unfortunately we don't treat aspect related to CubeMX tool in our GitHub repositories. All aspects related to Cube MX tool are not handled at GitHub level. They are rather treated at ST Community level, precisely within the Cube MX dedicated [section](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus).

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem (namely the Cube MX tool), please allow me to close it.

Thank you for your contribution. We are looking forward to reading from you again.

With regards, 

Thank you for your comprehension.

With regards,

closed 2021-08-06 by RKOUSTM